amaranth-lang/amaranth-boards #58: Add release

FFY00, 2020-03-27:
Hey, would it be possible to tag a release so that migen-boards can be properly distributed?

FFY00, 2020-03-27:
It would also be great to have releases for the other components, let me know if you want me to open issues there.

whitequark, 2020-03-28:
The plan for nmigen-boards is that each commit to master would automatically push a tag and a PyPI release, since there is no such thing as "stable boards". Either you track master, or you pin a specific commit / version.

This isn't currently implemented, and until it's implemented I don't have any plans to push individual releases manually. Feel free to package HEAD.

FFY00, 2020-03-28:
> This isn't currently implemented, and until it's implemented I don't have any plans to push individual releases manually. Feel free to package HEAD.

ACK

Would it make sense to sync up with nmigen releases? This only makes sense if there is a possibility for the API to break.

The other components (stdio and soc) are a bit different, what do you want to do there?

whitequark, 2020-03-28:
> Would it make sense to sync up with nmigen releases? This only makes sense if there is a possibility for the API to break.

At any given moment, nmigen-boards is supposed to depend on the range of nmigen versions it's guaranteed to be compatible with. (Sometimes that might not be true--it's not easy to get pip to perform minimal version resolution--but that's the intent.)

> The other components (stdio and soc) are a bit different, what do you want to do there?

At the moment those components are quite immature and it's too early to talk about releasing them. Let's discuss it once it actually makes sense.

FFY00, 2020-03-28:
> At the moment those components are quite immature and it's too early to talk about releasing them. Let's discuss it once it actually makes sense.

Sure, please ping me if you remember.

closed 2020-03-28 by FFY00