amaranth-lang/amaranth-boards #61: Spork Errors

zignig, 2020-05-07:
Building the ideal_spork I came across an cool error block.

command 

python -m ideal_spork --dumpall --force

Errors:

Ideal Spork
20200507 21:28:46 - ERROR - upduino_v2    __main__ - line 32 - (<module>)
20200507 21:28:46 - ERROR -     Resource SB_HFOSC#0 does not exist        __main__ - line 33 - (<module>)
20200507 21:28:46 - ERROR - de10_lite     __main__ - line 32 - (<module>)
20200507 21:28:46 - ERROR -     Resource clk50#0 does not exist           __main__ - line 33 - (<module>)
20200507 21:28:46 - ERROR - tinyfpga_ax1          __main__ - line 32 - (<module>)
20200507 21:28:46 - ERROR -     Can't instantiate abstract class TinyFPGAAX1Platform with abstract methods resources      __main__ - line 33 - (<module>)
20200507 21:28:46 - ERROR - tinyfpga_ax2          __main__ - line 32 - (<module>)
20200507 21:28:46 - ERROR -     Can't instantiate abstract class TinyFPGAAX2Platform with abstract methods resources      __main__ - line 33 - (<module>)
20200507 21:28:46 - ERROR - zturn_lite_z010       __main__ - line 32 - (<module>)
20200507 21:28:46 - ERROR -     Platform 'ZTurnLiteZ010Platform' does not define a default clock          __main__ - line 33 - (<module>)
20200507 21:28:46 - ERROR - upduino_v1    __main__ - line 32 - (<module>)
20200507 21:28:46 - ERROR -     Resource SB_HFOSC#0 does not exist        __main__ - line 33 - (<module>)
20200507 21:28:46 - ERROR - zturn_lite_z007s      __main__ - line 32 - (<module>)
20200507 21:28:46 - ERROR -     Platform 'ZTurnLiteZ007SPlatform' does not define a default clock         __main__ - line 33 - (<module>)

Patch can be built if needed.

rroohhh, 2020-05-07:
So `ZTurnLiteZ010Platform` and `ZTurnLiteZ007SPlatform` don't really have a useful default clock. There is no oscillator connected to any of the fpga pins. 

As clock source usually one of the clocks coming from the ARM side is used (one of the `FCLK`s). The main problem with adding them as a default clock is, that their clock frequency is programmable and thus generally unknown without extra input from the user. 
I think the default is 100 MHz, but I think I would rather have users specify the frequency they are using explicitly to avoid hard to find bugs.

zignig, 2020-05-07:
Indeed , none of the above do. 

When I define the resources of the tinyfpga_a* they do the same thing.

20200507 22:23:15 - ERROR - tinyfpga_ax1          __main__ - line 45 - (<module>)
20200507 22:23:15 - ERROR -     Platform 'TinyFPGAAX1Platform' does not define a default clock    __main__ - line 46 - (<module>)
20200507 22:23:15 - ERROR - tinyfpga_ax2          __main__ - line 45 - (<module>)
20200507 22:23:15 - ERROR -     Platform 'TinyFPGAAX2Platform' does not define a default clock    __main__ - line 46 - (<module>)

Just wanted to point out a __weridness__.

zignig, 2020-05-07:
see https://github.com/nmigen/nmigen-boards/pull/62

whitequark, 2020-05-08:
All of the other errors other than the ones fixed in #62 are caused by your code not accounting for the fact that a default clock or its frequency might be missing.

closed 2020-05-08 by whitequark

zignig, 2020-05-08:
interestingly , with the boards that have an internal clock just asking for the default_clk_frequency attribute causes the resource lookup to fail.

20200508 15:23:18 - ERROR - upduino_v1    __main__ - line 53 - (<module>)
20200508 15:23:18 - ERROR -     Resource SB_HFOSC#0 does not exist        __main__ - line 54 - (<module>)
Traceback (most recent call last):
  File "/opt/FPGA/ideal-spork/ideal_spork/__main__.py", line 47, in <module>
    devices = map_devices(bi)
  File "/opt/FPGA/ideal-spork/ideal_spork/builder/map_board.py", line 100, in map_devices
    clock = check_clock(board_instance)
  File "/opt/FPGA/ideal-spork/ideal_spork/builder/map_board.py", line 65, in check_clock
    if hasattr(board_instance,'default_clk_frequency'):
  File "/home/zignig/.local/lib/python3.6/site-packages/nmigen-0.2-py3.6.egg/nmigen/build/plat.py", line 44, in default_clk_frequency
    constraint = self.default_clk_constraint
  File "/home/zignig/.local/lib/python3.6/site-packages/nmigen-0.2-py3.6.egg/nmigen/build/plat.py", line 40, in default_clk_constraint
    return self.lookup(self.default_clk).clock
  File "/home/zignig/.local/lib/python3.6/site-packages/nmigen-0.2-py3.6.egg/nmigen/build/res.py", line 58, in lookup
    .format(name, number))
nmigen.build.res.ResourceError: Resource SB_HFOSC#0 does not exist