amaranth-lang/amaranth-boards #74: DDR pins and *_n signals

jeanthom, 2020-07-07:
Hi,

For some peripherals (eg. DDR memory) there are inverted pins (RAS_N, CAS_N, etc. in DDR3) that are configured as PinsN. The issue with PinsN is that it gets an inverter inserted between the TRELLIS_IO instance (on ECP5 platform) and the .o exposed to the user which prevents the use of DDR primitives (*ERROR: ODDRX2F 'ddrphy.U$$19' Q output must be connected only to a top level output*).

Assuming this issue isn't vendor-specific, should we declare those resources with the "*_n" naming style and expect the user to do the inversion themselves? Or should we keep it as-is and request the resource as raw IO?

whitequark, 2020-07-07:
> The issue with PinsN is that it gets an inverter inserted between the TRELLIS_IO instance (on ECP5 platform) and the .o exposed to the user which prevents the use of DDR primitives

Let's discuss this on IRC next time we're both online, that would be more productive I think.

whitequark, 2020-07-13:
The question was resolved on IRC: https://freenode.irclog.whitequark.org/nmigen/2020-07-08#1594197820-1594198832

closed 2020-07-13 by whitequark