amaranth-lang/amaranth-boards #141: Reason for duplication toolchain_program function

hellow554, 2021-02-05:
I'm playing around with a intel/altera based platform and found it very easy to define my own platform, that's amazing! :heart:  (I'm probably doing a PR in the near future for it).

What wonders me is that (at least the IntelPlatform deivces) share (almost) the same programming code. The only difference I found is that they appear in a different position in the JTAG chain.

What is the reason for not having [the](https://github.com/nmigen/nmigen-boards/blob/a35d870a994c2919116b2c06166dc127febb1512/nmigen_boards/chameleon96.py#L94-L100) [code](https://github.com/nmigen/nmigen-boards/blob/a35d870a994c2919116b2c06166dc127febb1512/nmigen_boards/de0_cv.py#L98-L102) [in](https://github.com/nmigen/nmigen-boards/blob/a35d870a994c2919116b2c06166dc127febb1512/nmigen_boards/mister.py#L123-L129) (...) the (`IntelPlatform`)[https://github.com/nmigen/nmigen/blob/master/nmigen/vendor/intel.py] class? 

I also would like to see the `quartus_pgm` program in the required tools list if `do_program` is `True`.

What are your thoughts about it? I would like to take that as an approach to commit to this project (:

whitequark, 2021-02-05:
The problem with using a common `toolchain_program` function is that almost every board is unique in *some* way, and so every change to the common function would, to be robust, require re-testing of every Intel board. That's unfeasible.

Also, not everything programs through Quartus.

hellow554, 2021-02-05:
It seems that you got me wrong.
I don't mean that **every** board should be have the same `toolchain_program` function, only the board that share the same programming procedure, e.g. all `IntelPlatform` tools (which all of them require the `quartus_pgm` tool ;)

Does that make more sense to you?

whitequark, 2021-02-05:
Not all Intel FPGA boards inherently require `quartus_pgm`. The ones we currently have might, but there is no such inherent requirement as boards can and do have custom programming tools.

hellow554, 2021-02-05:
Okay, you're right, that may be a possibility. So don't require `quartus_pgm` as a tool.

But for the second part, what do you think about promiting the `toolchain_program` method to the `IntelPlatform` class (and let possible future board override that method to have their own logic)

whitequark, 2021-02-05:
That would be a liability for the reason I have described: right now `toolchain_program` is entirely self-contained and can be tested just once when the board is added, and with your proposed design, that won't be possible.

hellow554, 2021-02-05:
I'm not entirely getting your point here, but I'm ok with it.
For me it is that 8 boards do (almost) exactly the same when programming, so we have that code 8 times in our database.

As said, if you think that's the right way to go because of the reasons you listed above, I'm fine with it.
Thanks for the clarification :)

closed 2021-02-05 by hellow554