amaranth-lang/amaranth-boards #145: Add support for Nandland Go Board

nono2357, 2021-03-03:
Here are the board details:
https://www.nandland.com/goboard/introduction.html

Excellent board for beginners and open source projects.

Thank you.

whitequark, 2021-03-03:
We do not, in general, accept requests for new board definitions through issues in this repository. I suggest implementing support for this board yourself or finding someone to implement it, and submitting it as a pull request.

StephanvanSchaik, 2021-05-29:
I ordered the Nandland Go board, and I will gladly write up and test the board definition for it once it arrives. Given that shipping might take about 4-5 business days, I am estimating that to be for the next weekend.

kivikakk, 2023-03-04:
Just a heads up, this issue can be closed since the merge of #164.

closed 2023-03-04 by whitequark