amaranth-lang/amaranth-boards #176: Nexys4DDR: I/O standard should be different for some switches

neuschaefer, 2021-08-01:
On the Nexys4DDR board, two of the 16 slide switches (SW8/T8, SW9/U8) are connected to 1.8V rather than 3.3V:

[image]

This is currently not reflected in the [board definition](https://github.com/nmigen/nmigen-boards/blob/master/nmigen_boards/nexys4ddr.py#L24), but probably should be.

whitequark, 2021-08-01:
Would you be willing to send a PR? Here is an example of how to use `SwitchResources` with two different IO standards: https://github.com/nmigen/nmigen-boards/blob/6bbd2dd89f78f38f11ec497e34d4cddf8aae88dc/nmigen_boards/ecp5_5g_evn.py#L58-L61

neuschaefer, 2021-08-01:
Yes, will do. Thanks for the guidance.

PR #179 (2021-08-13): nexys3ddr: Fix I/O voltage for SW8 and SW9
Fixes #176.
closed 2021-08-13 by whitequark