amaranth-lang/amaranth-boards #189: Allow to name SplitResources based resources (like LEDResources, ButtonResources...)

hansfbaier, 2021-12-19:
Currently it is not possible to create those with a different name than their default name,
which is hardwired.

bl0x, 2022-01-23:
I ran into the same problem when trying to run the tests.
I left a note on how one might be able to solve it here:

https://github.com/bl0x/amaranth-boards/blob/c58f4d1a5354b82d9e9f132ecac64561918980cc/amaranth_boards/genesys2.py#L37

bl0x, 2022-01-23:
I made a proposal on how to solve this:

https://github.com/amaranth-lang/amaranth-boards/pull/192

hansfbaier, 2022-01-23:
@bl0x Very cool, thanks!

whitequark, 2022-02-09:
The `LEDResources`, `ButtonResources` and other similar helpers are intended to simplify a common case when defining resources, and to nudge the board definition file author towards our conventions. They are not in any way *required* to define LED, button, etc. resources; so for the relatively rare case where a resource with a custom name is needed, they can be expanded in the board definition file.

See also https://github.com/amaranth-lang/amaranth-boards/pull/192#issuecomment-1033359434.

closed 2022-02-09 by whitequark

bl0x, 2022-02-09:
> The `LEDResources`, `ButtonResources` and other similar helpers are intended to simplify a common case when defining resources, and to nudge the board definition file author towards our conventions. They are not in any way _required_ to define LED, button, etc. resources; so for the relatively rare case where a resource with a custom name is needed, they can be expanded in the board definition file.
> 
> See also [#192 (comment)](https://github.com/amaranth-lang/amaranth-boards/pull/192#issuecomment-1033359434).

Perhaps this can be mentioned in the documentation?

whitequark, 2022-02-09:
Sure, once there's actual documentation for the board DSL. (This will likely come after the *redesign* of the board DSL, so...)