amaranth-lang/amaranth-boards #197: Please add iCESugar.Pro support

gromas, 2022-03-12:
[icesugar_pro.zip](https://github.com/amaranth-lang/amaranth-boards/files/8237100/icesugar_pro.zip)

board has some more resources but this is basics for tests

josuah, 2023-03-17:
```python
import subprocess

from amaranth.build import *
from amaranth.vendor.lattice_ecp5 import *
from .resources import *

__all__ = ["ICESugarProPlatform"]

class ICESugarProPlatform(LatticeECP5Platform):
    device      = "LFE5U-25F"
    package     = "BG256"
    speed = "6"
    default_clk = "clk25"
    default_rst = "rst"
    resources   = [
        Resource("clk25", 0, Pins("P6", dir="i"), 
                 Clock(25e6), Attrs(IO_STANDARD="LVCMOS33")),
        Resource("rst", 0, Pins("L14",dir="i"), Attrs(IO_TYPE = "LVCMOS33")),

        RGBLEDResource(0, r="B11", g="A11", b="A12", attrs=Attrs(IOSTANDARD="LVCMOS33"))

    ]

    def toolchain_program(self, products, name):
        with products.extract("{}.bin".format(name)) as bitstream_filename:
            subprocess.check_call(["icesprog", bitstream_filename])

if __name__ == "__main__":
    from .test.blinky import *
    ICESugarProPlatform().build(Blinky(), do_program=True)
```

josuah, 2023-03-17:
@gromas thank you, this helped consolidating #198