amaranth-lang/amaranth-boards #200: CI is broken

hansfbaier, 2022-04-15:
All tests terminate with this error:
```
ImportError: cannot import name 'soft_unicode' from 'markupsafe' (/home/runner/work/amaranth-boards/amaranth-boards/.eggs/MarkupSafe-2.1.1-py3.10-linux-x86_64.egg/markupsafe/__init__.py)

```

MidnightSkylight, 2022-04-16:
Yes, this is a known issue, the issue in Amaranth itself was patched with https://github.com/amaranth-lang/amaranth/issues/688, while the issue with boards and testing in general is related to https://github.com/amaranth-lang/amaranth-boards/issues/128. There aren't even any tests defined in the repository so I don't see this is a practical issue and it cannot be solved without addressing #128 anyway.

sporniket, 2022-10-06:
@whitequark , do you have a preference between between my PR #205 or my ticket [amaranth#718](https://github.com/amaranth-lang/amaranth/issues/718) ? So that I can close the one that you don't like ?

No problem if you don't like both, but I feel that this problem is stalling a lot of PRs.

whitequark, 2022-10-06:
I'm currently in the process of moving to another country which is taking a lot of time and effort. Once I'm done (or possibly earlier) I'm going to cut an Amaranth release which will solve this issue.

sporniket, 2022-10-06:
ok, take care

whitequark, 2023-02-03:
CI is fixed now. There's a workaround for downgrading MarkupSafe with Amaranth 0.3.

closed 2023-02-03 by whitequark