amaranth-lang/amaranth-boards #207: Upgrading from nMigen and cannot get amaranth-boards installed

JarrettBillingsley, 2022-11-26:
I have a little project I'm trying to get up-to-date and I've run into a weird issue. This repo is very clearly full of board definitions including the one I was using with nMigen, but when I try to install it with pip in my virtualenv, I get... nothing?

```bash
(venv) $ pip install amaranth-boards
Collecting amaranth-boards
  Downloading amaranth_boards-0-py3-none-any.whl (1.1 kB)
Installing collected packages: amaranth-boards
Successfully installed amaranth-boards-0

(venv) $ ls venv/lib/python3.9/site-packages/
total 4
drwxr-xr-x  5 me staff  160 Nov 26 17:04 _distutils_hack
drwxr-xr-x 20 me staff  640 Nov 26 17:11 amaranth
drwxr-xr-x  7 me staff  224 Nov 26 17:11 amaranth_yosys
drwxr-xr-x 30 me staff  960 Nov 26 17:11 jinja2
...
drwxr-xr-x  8 me staff  256 Nov 26 17:35 amaranth_boards-0.dist-info
...
```

There is a `amaranth_boards-0.dist-info` directory in site-packages but there is no `amaranth_boards` directory. 

Am I missing something? I know this section of the docs is still a TODO: https://amaranth-lang.org/docs/amaranth/latest/install.html#installing-board-definitions so I don't know what else I'm missing here.

JarrettBillingsley, 2022-11-27:
I was able to get it manually installed from the git repo:

```bash
pip install git+https://github.com/amaranth-lang/amaranth-boards.git#egg=amaranth-boards
```

but idk if that's what I'm supposed to do.

whitequark, 2023-02-03:
Yup, at the moment this is appropriate. Releasing amaranth-boards is tracked at #128.

closed 2023-02-03 by whitequark